iotsound/iotsound #604: Intel NUC Build failing

Chumba, 2022-11-16:
**Describe the problem**
I'm unable to deploy on NUC, it seems like audio-amd64 and bluetooth amd-64 had some links change today and balena-sound needs to be updated (I'm a complete noob here so forgive me if my conclusions are erroneous)

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to deploy balena-sound on a NUC based client
4. See error

**Expected behavior**
balena-sound successfully deploys

**Screenshots**
[image]

**Device information**
 - Board (hardware info): NUC
 - Device type (as seen in balenaCloud): Intel NUC
 -  balenaOS version: balenaOS 2.105.31+rev1

Build Overview:
[Info]              Starting build for testFleet, user g_bryan_charalambous
[Info]              Dashboard link: https://dashboard.balena-cloud.com/apps/1990690/devices
[Info]              Building on x64_01
[Info]              No suitable previous release for caching, skipping cache pull
[Info]              Starting to build hostname
[Info]              Starting to build multiroom-client
[Info]              Starting to build audio
[Info]              Starting to build bluetooth
[Info]              Starting to build airplay
[Info]              Starting to build spotify
[Info]              Starting to build multiroom-server
[Info]              Starting to build sound-supervisor
[Success]           Successfully built service hostname
[Success]           Successfully built service spotify
[Success]           Successfully built service airplay
[Success]           Successfully built service multiroom-client
[Success]           Successfully built service multiroom-server
[Success]           Successfully built service sound-supervisor
[Info]              Uploading images
[Success]           Successfully uploaded images
[Error]             Some services failed to build:
[Error]               Service: audio
[Error]                 Error: pull access denied for bh.cr/balenablocks/audio-amd64, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
[Error]               Service: bluetooth
[Error]                 Error: pull access denied for bh.cr/balenablocks/bluetooth-amd64, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
[Info]              Built on x64_01
[Error]             Not deploying release.

Chumba, 2022-11-16:
Relevant PRs from the affected projects

https://github.com/balena-labs-projects/bluetooth/pull/29
https://github.com/balena-labs-projects/audio/pull/102

Chumba, 2022-11-16:
Sorry fat fingered the closed button

nucleardreamer, 2022-11-16:
@Chumba you caught it alright. We are in the middle of transitioning some containers between different repos and this project was a little lagged to get updated there. Should be resolved now!

closed 2022-11-16 by nucleardreamer

Chumba, 2022-11-17:
Thanks for the update! Having lots of fun with balena, really cool stuff you guys got going on.