iotsound/iotsound #607: Upgrade Shairport Sync to version 4.1

fabienheureux, 2022-11-25:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

maggie0002, 2022-11-27:
Thanks for taking a look at this, it is very much appreciated. 

You can clone this repository, make amendments to it, and then push it to your own fleet for testing: https://www.balena.io/docs/learn/deploy/deployment/#balena-push

fabienheureux, 2022-11-29:
Thank for your help, I did not think of checking Balena's support pages...
I will try this as soon as I can !

Neustradamus, 2023-01-24:
To follow!

maggie0002, 2023-01-25:
Closed by: https://github.com/balena-labs-projects/balena-sound/pull/621

closed 2023-01-25 by maggie0002