iotsound/iotsound #608: Push to x86 : image not available am64

YossefAboukrat, 2022-11-28:
Hello,
I deploy to a generic x86 device and it fails to builder . Can't find amd64 images.

To reproduce do a balena deploy to a generic x86 device

it shows me for each docker-compose services this errors : 

`pull access denied for bh.cr/balenablocks/bluetooth-amd64, repository does not exist or may require 'docker login': denied: requested access to the resource is denied`

In device : 
- odyssey x86
- balena os generic x86

maggie0002, 2022-11-29:
Check you have the latest code from this repository. The container it is trying to pull has changed: https://github.com/balena-labs-projects/balena-sound/commit/ec21ca9dcd5da43fbab2559bea165b7ceeb6b1cc

YossefAboukrat, 2022-11-29:
Brilliant, its walking! thank you !
one last thing, I had to switch to standalone instead of multi-room otherwise it crashed, it made this error:

  multiroom-server --- Multiroom ---
  multiroom-server Starting multiroom service with settings:
  multiroom-server - Multiroom mode: bash
  multiroom-server - Snapcast version: v0.26.0
  multi-room-server Starting multi-room-server...
  multiroom-server 2022-11-29 07-35-51.261 [Error] (Avahi) Failed to create client: Daemon not running
  multiroom-server ALSA lib pcm_pulse.c:752:(pulse_prepare) PulseAudio: Unable to create stream: No such entity
  multiroom-server
  multiroom-server 2022-11-29 07-35-51.274 [Error] (Snapserver) Exception: Can't set hardware parameters: I/O error

but with standalone it works perfectly.

maggie0002, 2022-11-29:
If the other issue persists check to see if there is an issue open already, and if not then you can open a new one.

closed 2022-11-29 by maggie0002