iotsound/iotsound #611: Shows Status: Failed in Releases. Build Failed. Unable to deploy Balena Sound to fleet.

RubixStars, 2022-12-02:
Shows Status: Failed in Releases. Build Failed. Unable to deploy Balena Sound to fleet. 

Steps to reproduce the behavior:
1. Click on 1 click deploy button
2. Deploy to a fleet containing raspberry pi 3
3. Go to releases and wait.
4. Shows Status: Failed

Expected for all services to run seamlessly

**Screenshots**
[image]
[image]
[image]
[image]

**Device information**
 - Board Raspberry Pi 3B+
 - Device type Raspberry Pi 3 
 - balenaOS version: balenaOS 2.107.10+rev3

maggie0002, 2022-12-03:
@andrewn may have an idea?

andrewn, 2022-12-03:
Is there a reason why the Balena builder wouldn't be able to fetch https://github.com/rust-lang/crates.io-index?

I may be misunderstanding the cargo error though...

```
Step 10/17 : RUN cargo build     --release     --features alsa-backend,pulseaudio-backend     --verbose
 ---> Running in e2531cf3a38f
 
   
Updating
 crates.io index

error
:
 
failed to get `base64` as a dependency of package `librespot v0.4.2 (/app/librespot-src)`

Caused by:

  failed to fetch `https://github.com/rust-lang/crates.io-index`

Caused by:

  unexpected data at the end of the pack; class=Indexer (15)
  ```

andrewn, 2022-12-03:
Running `balena push gh_andrewn/balena-sound-master-andrewn`

Fails with an out-of-memory error...

```
[spotify]           warning
[spotify]           :
[spotify]            spurious network error (2 tries remaining): failed to mmap. Could not write data: Out of memory; class=Os (2)
[spotify]
```

maggie0002, 2022-12-05:
@RubixStars, are you using the standard Raspberry Pi 3 image, or the 64bit one? Does it fail on both?

closed 2022-12-05 by maggie0002

andrewn, 2022-12-05:
Can you try and build a new release? It should have been fixed in PR #613.

saveriogzz, 2022-12-05:
I had the same issue and I confirm that the build now ends correctly. Thanks!

PR #613 (2022-12-05): Use a pre-built librespot image
I have no idea why the `librespot` building isn't working (implemented in PR #610) causing issues #611 and #612.

### Option 1:

I think we should switch to the pre-built images of v0.4.2.

[There's a list of architectures here](https://hub.docker.com/r/andrewn/librespot/tags).

[I used this script](https://github.com/andrewn/librespot-docker/blob/andrewn/librespot-v0.4.2/scripts/build-pulseaudio.sh) and have been running on a Pi model 3 A+ for a few months.

### Option 2:

Open a PR to revert commit 9f8c623c7c6778e19ce20a99e4b0029bb85e316b. 

@maggie0002 it's u [... cut]