iotsound/iotsound #612: Spotify module fails with 101 exit code

MrJynxZ, 2022-12-03:
**Describe the problem**
A clear and concise description of what the problem is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...' after adding a device with a Raspberry Pi 4 using the latest (recommended) and previous revision.
2. Click on '....' the device shows up on the balena dashboard
3. Scroll down to '....' when adding audio-sound to it, the compilation fails on the Spotify module with a 101 returned code
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device information**
 - Board (hardware info): [e.g. Raspberry Pi 4B- 8GB]
 - Device type (as seen in balenaCloud): [e.g. Raspberry Pi 4 (using 64bit OS)]
 - balenaOS version: [
 - balena-cloud-balena-sound-raspberrypi4-64-2.107.10+rev3-v14.4.4.img
 - balena-cloud-balena-sound-raspberrypi4-64-2.106.2-v14.3.0.img
 - both failed the same way
[balena-sound_release-log_spotify-512ce43.txt](https://github.com/balena-labs-projects/balena-sound/files/10147349/balena-sound_release-log_spotify-512ce43.txt)
]
all other modules passed the compilation

**Additional context**
Add any other context about the problem here.

elkir0, 2022-12-03:
Same With Rpi3 B+ 

[Success]           Successfully built service hostname
[Success]           Successfully built service audio
[Success]           Successfully built service multiroom-client
[Success]           Successfully built service multiroom-server
[Success]           Successfully built service bluetooth
[Success]           Successfully built service sound-supervisor
[Info]              Uploading images
[Success]           Successfully uploaded images
[Error]             Some services failed to build:
[Error]               Service: spotify
[Error]                 Error: The command '/bin/sh -c cargo build     --release     --features alsa-backend,pulseaudio-backend     --verbose' returned a non-zero code: 101
[Info]              Ran locally emulated build
[Error]             Not deploying release.

btraquair, 2022-12-03:
I am seeing the exact same behavior with raspberry pi v1

btraquair, 2022-12-05:
I've just tried to deploy from balena. Spotify built and installed properly!

closed 2022-12-06 by maggie0002