iotsound/iotsound #614: Audio not coming out jack 3.5mm on RPi4

saveriogzz, 2022-12-13:
**Describe the problem**
Using a RaspberryPi4 and setting the env variable `AUDIO_OUTPUT` to either `RPI_AUTO` or `RPI_HEADPHONES`. Does not make the audio coming out of the RPi4's 3.5mm jack.
Using the latest commit e3098287cb9b0b866e528251b424e388e3fbbb6a

**To Reproduce**
Steps to reproduce the behavior:
1. Set the device as a `STANDALONE` and set the env variable `AUDIO_OUTPUT` to either of the two options above.
2. Try to reproduce music
3. Check result

**Expected behavior**
I would expect the audio to work since I've tested it with RaspberryPi OS and the jack works so it's not a hardware issue.

**Device information**
 - Raspberry Pi 4 64bit
 - Raspberry Pi 4 (using 64bit OS)
 - balenaOS 2.107.10+rev3

AnnoyedPlatypus, 2022-12-28:
I'm experiencing the same issue - everything seems to work, except the audio. Other than the above, I note the following in the log info shown in the Balena console.

```
28.12.22 19:11:21 (+1300)  audio  E: [pulseaudio] module-alsa-card.c: Failed to find a working profile.
28.12.22 19:11:21 (+1300)  audio  E: [pulseaudio] module.c: Failed to load module "module-alsa-card" (argument: "device_id="0" name="platform-fef00700.hdmi" card_name="alsa_card.platform-fef00700.hdmi" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=yes card_properties="module-udev-detect.discovered=1""): initialization failed.
```

AnnoyedPlatypus, 2022-12-28:
Update, I have cleared the "vc4-kms-v3d" value from the "Define DT overlays" option in the Device Configuration list. After a reboot, and restarting the airplay service for some reason, the audio from the 3.5mm jack is working.

saveriogzz, 2023-01-12:
@AnnoyedPlatypus just got the opportunity to test this but nope, still not working!

closed 2023-01-15 by saveriogzz