iotsound/iotsound #623: Spotify fails to build for amd64

lumbarthought, 2023-01-23:
Deploying balenasound to a generic x86_64 or Intel NUC fails due to Spotify not building. Error log shows:

`[Error]             Some services failed to build:
[Error]               Service: spotify
[Error]                 Error: no matching manifest for linux/amd64 in the manifest list entries`

lumbarthought, 2023-01-23:
From the spotify build log

```
Step 1/4 : FROM andrewn/librespot:0.4.2-pulseaudio
no matching manifest for linux/amd64 in the manifest list entries
```

maggie0002, 2023-01-23:
Thanks for reporting this.

@andrewn, it does look like the amd64 image version hasn't built: https://hub.docker.com/r/andrewn/librespot/tags

maggie0002, 2023-01-23:
@andrewn, this may be an easier approach: https://github.com/andrewn/librespot-docker/pull/1

If we merge that in to the repo then we can build the containers here on GitHub rather than need to push to Docker Hub. We should be able to merge that, then add a tag to the latest commit called `ghcr` and it will start a build. Assuming they all build ok, then we can point balenaSound at those new GHCR images instead.

andrewn, 2023-02-08:
@maggie0002 Thanks for opening the PR! I managed to get the builds working and [the containers are here ](https://github.com/andrewn/librespot-docker/pkgs/container/librespot-docker).

I'm travelling until the end of February so won't be able to test on a real Pi until then.

jkr442, 2023-02-19:
Would there be a workaround possible for those of us trying to set up devices in the meantime?

andrewn, 2023-02-27:
[I created the `amd64` image here](https://hub.docker.com/layers/andrewn/librespot/0.4.2-pulseaudio-amd64/images/sha256-43eb03de4b66cd4ed8e9bf6f28b5912b6718d6ed7dc5c0fad254dbee9d0bbc37?context=repo). 

It should install if you try the build again. Want to give it a go @jkr442?

jkr442, 2023-02-27:
Success!
Thanks