iotsound/iotsound #626: Spotify cache and normalization customization variables don't work if their value is not defined

laytonhayes, 2023-01-25:
The configuration variables **SOUND_SPOTIFY_DISABLE_NORMALISATION** and **SOUND_SPOTIFY_ENABLE_CACHE** only work if the value is defined. It is not clear in the support docs that _any_ value has to be declared.

**Additional context**
In /plugins/spotify/start.sh these variable values are checked for null or not null using -z. If the variable is set but no value declared it functions as if there is no variable at all.
```
if [[ -z "$SOUND_SPOTIFY_DISABLE_NORMALISATION" ]]; then
  set -- "$@" \
    --enable-volume-normalisation
fi 
```
**Suggestions**
Update the support docs to reflect that a value has to be set to something OR check to see if that variable exist, even if it's set to null

maggie0002, 2023-01-25:
Great catch. Do you think that is what could be happening here: https://github.com/balena-labs-projects/balena-sound/issues/620?

Would you be able to submit a PR?

laytonhayes, 2023-01-31:
@maggie0002 - I've created PR #628 for the Spotify variable fixes.

closed 2023-02-06 by maggie0002