iotsound/iotsound #629: Documentation about

maxoux, 2023-02-03:
Hi everyone !

**Is your feature request related to a problem? Please describe.**
As i just stumbled upon this repo, i'm interested in making a multiroom sound system based on raspberry pi's.
One of my prerequisite is to be as less dependant on third-party providers, which seem the case seeing the first lines of the READ.md : 

> There are many commercial solutions out there that provide functionality similar to balenaSound. Most of them though come with a premium price tag and are riddled with privacy concerns.

But as the first step is to subscribe to balenacloud which is, not only limit device  up to ten, seem to heavily operate on online connection on the raspberry pi clients; and i didn't see any documentation nor really understand the role of balenacloud in the process.

**Describe the solution you'd like**
Is it possible to know if indeed balenacloud is mandatory (which doesn't seems to be the case) and more importantly, any documentation or guide to locally setup a fleet would be appreciated.

**Additional context**
I would like to note that i indeed searched and found some partial informations on ols forum (from 2020 to 2022), but a lot of mentionned links are dead, and the few documentation i found (mainly [this](https://docs.balena.io/reference/OS/overview/) and [that](https://docs.balena.io/learn/develop/local-mode/#local-mode-requirements) and [this issue](https://github.com/balena-labs-projects/balena-sound/issues/558) ) either have only some mention about local setup or straight up need a cloud account for this.

maggie0002, 2023-02-06:
You could download the balenaOS, and then push the code to the device locally: https://docs.balena.io/learn/deploy/deployment/#balena-push

TorbenHammes, 2025-05-11:
Hi @maxoux,

1. Install **balena CLI**:
   https://github.com/balena-io/balena-cli

2. Build all services locally:
   ```bash
   git clone https://github.com/balena-io-experimental/balena-sound.git
   cd balena-sound
   balena build --deviceType <yourDeviceType> .

`balena deploy` doesn’t work here since we’re not using a cloud fleet. After building with `balena build`, all images are available locally:

```bash
docker image ls
````

You can start them directly with Docker Compose without rebuilding:

```bash
# Retag images so Compose can find them
for svc in sound-supervisor spotify bluetooth audio hostname multiroom-server airplay multiroom-client; do
   docker tag balena-sound_$svc:latest balena-sound-$svc:latest
done

# Start services without rebuilding
docker compose up -d --no-build
```

This runs balena-sound entirely locally—no balenaCloud or balenaOS required.