iotsound/iotsound #634: Balena Sound keeps waiting for supervisor

Hintenhaus04, 2023-03-07:
i am running Balena sound on a raspberry pi 3A+, but i can't connect via anyting. everyting keeps disconnecting after a few seconds. no airplay, spotify connect or bluetooth. 
i am running all services but every service keeps waiting for the sound-supervisor. even when i kill the sound-supervisor they are still waiting for a ip-adress that doesn't exist in my network.

this is what i got
 audio  Waiting for sound supervisor to start at 172.17.0.1:80
 multiroom-client  Waiting for sound supervisor to start at 172.17.0.1:80
Restarting service 'sound-supervisor sha256:6d607fcb4205bdbcd29d12576d576a7cdd180c830674efaa190074ccba782049'
 sound-supervisor  standard_init_linux.go:219: exec user process caused: exec format error
Service exited 'sound-supervisor sha256:6d607fcb4205bdbcd29d12576d576a7cdd180c830674efaa190074ccba782049'
 multiroom-client  Waiting for sound supervisor to start at 172.17.0.1:80
 multiroom-server  Waiting for sound supervisor to start at 172.17.0.1:80
 audio  Waiting for sound supervisor to start at 172.17.0.1:80
 multiroom-client  Waiting for sound supervisor to start at 172.17.0.1:80

 anyone any suggestions?

robert-go, 2023-03-09:
same with me

Nico239SF, 2023-03-17:
same thing

nbwdk, 2023-03-19:
I am getting the same error. The Container keeps restarting.

Error response from daemon: Container 1908008c48508d8c38b6390d0dbe074735f97a849372da190cd8b699cc197fc1 is restarting, wait until the container is running

tangorri, 2023-03-25:
same for me

cgpuglie, 2023-03-31:
Same issue here

meierdan, 2023-04-20:
I have the same with latest balena-sound update
root@balena:~# balena logs sound-supervisor_6454329_2559361_5fce214260f3dc89830b60cdf1b3c600
standard_init_linux.go:219: exec user process caused: exec format error
standard_init_linux.go:219: exec user process caused: exec format error
standard_init_linux.go:219: exec user process caused: exec format error
standard_init_linux.go:219: exec user process caused: exec format error
standard_init_linux.go:219: exec user process caused: exec format error
standard_init_linux.go:219: exec user process caused: exec format error
standard_init_linux.go:219: exec user process caused: exec format error
standard_init_linux.go:219: exec user process caused: exec format error

af-device, 2023-05-13:
I got the same issue with a Raspberry Pi 3B

meierdan, 2023-05-13:
I found this solution
modify in core/sound-supervisor/Dockerfile.template

FROM balenalib/%%BALENA_ARCH%%-node:14 as build

WORKDIR /usr/src
COPY core/sound-supervisor/package*.json ./
RUN npm install

#dev-cmd-live=npm run livepush

COPY core/sound-supervisor/src ./src
COPY core/sound-supervisor/tsconfig.json .
RUN npm run build
COPY VERSION .

#FROM node:14-alpine3.14 (replaced by following line tho get proper architecture
FROM balenalib/%%BALENA_ARCH%%-alpine-node:14

WORKDIR /usr/src

COPY core/sound-supervisor/package*.json ./
COPY --from=build /usr/src/build ./build
COPY VERSION .

RUN npm install --production

CMD [ "npm", "start" ]

rorgeiir, 2023-05-18:
For anyone not quite technical enough for GitHub like me, I've created some instructions for applying the fix above over on the other issue comments thread #637

alanb128, 2023-06-05:
A fix for this has been merged.

closed 2023-07-01 by alanb128