iotsound/iotsound #636: ALSA bug

avassdal, 2023-03-23:
**Describe the problem**
Audio not working at all on RPi 4B.

ALSA reporting: ALSA woke us up to write new data to the device, but there was actually nothing to write.

Using headphone jack as output.

**To Reproduce**
Steps to reproduce the behavior:
Clean install of BalenaOS with BalenaSound deployed from Balena Hub

**Device information**
 - Board (hardware info): Raspberry Pi 4B 1.1
 - Device type (as seen in balenaCloud): Raspberry Pi 4 (using 64bit OS)
 - balenaOS version: balenaOS 2.113.18
 
**Additional context**
Logs:

> 

23.03.23 07:34:24 (+0100) Supervisor starting
23.03.23 07:33:46 (+0100)  audio  
23.03.23 07:33:46 (+0100)  audio  --- Audio ---
23.03.23 07:33:46 (+0100)  audio  Starting audio service with settings:
23.03.23 07:33:47 (+0100)  audio  - pulseaudio 15.0
23.03.23 07:33:47 (+0100)  audio  - Pulse log level: NOTICE
23.03.23 07:33:47 (+0100)  audio  - Default output: RPI_HEADPHONES
23.03.23 07:33:47 (+0100)  audio  - Default volume: 75%
23.03.23 07:33:47 (+0100)  audio  
23.03.23 07:33:47 (+0100)  audio  Detected audio cards:
23.03.23 07:33:47 (+0100)  audio  0 bcm2835-jack bcm2835_headpho-bcm2835Headphones
23.03.23 07:33:47 (+0100)  audio  
23.03.23 07:33:47 (+0100)  audio  
23.03.23 07:34:19 (+0100)  audio  Setting audio routing rules...
23.03.23 07:34:19 (+0100)  audio  Routing 'balena-sound.input' to 'balena-sound.output'.
23.03.23 07:34:19 (+0100)  audio  Routing 'balena-sound.output' to 'alsa_output.bcm2835-jack.stereo-fallback'.
23.03.23 07:34:19 (+0100)  audio  W: [pulseaudio] main.c: This program is not intended to be run as root (unless --system is specified).
23.03.23 07:34:21 (+0100)  audio  W: [pulseaudio] module-loopback.c: Cannot set requested sink latency of 66.67 ms, adjusting to 99.95 ms
23.03.23 07:34:21 (+0100)  audio  W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
23.03.23 07:33:45 (+0100)  spotify  Starting Spotify plugin...
23.03.23 07:33:45 (+0100)  spotify  Device name: Streamer U-Etg
23.03.23 07:33:45 (+0100)  spotify  [2023-03-23T06:33:45Z INFO  librespot] librespot 0.4.2 UNKNOWN (Built on 2022-08-01, Build ID: leSkOw9g, Profile: release)
23.03.23 07:33:45 (+0100)  spotify  [2023-03-23T06:33:45Z WARN  librespot] Could not initialise discovery: Setting up dns-sd failed: No such device (os error 19).
23.03.23 07:33:45 (+0100)  spotify  [2023-03-23T06:33:45Z ERROR librespot] Discovery is unavailable and no credentials provided. Authentication is not possible.
23.03.23 07:33:49 (+0100)  spotify  Starting Spotify plugin...
23.03.23 07:33:49 (+0100)  spotify  Device name: Streamer U-Etg
23.03.23 07:33:49 (+0100)  spotify  [2023-03-23T06:33:49Z INFO  librespot] librespot 0.4.2 UNKNOWN (Built on 2022-08-01, Build ID: leSkOw9g, Profile: release)
23.03.23 07:33:49 (+0100)  sound-supervisor  
23.03.23 07:33:49 (+0100)  sound-supervisor  > sound-supervisor@1.0.1 start /usr/src
23.03.23 07:33:49 (+0100)  sound-supervisor  > node build/index.js
23.03.23 07:33:49 (+0100)  sound-supervisor  
23.03.23 07:34:15 (+0100)  sound-supervisor  Sound supervisor listening on port 80
23.03.23 07:34:15 (+0100)  sound-supervisor  Error connecting to audio block - Retry failed: connect ECONNREFUSED 192.168.2.183:4317
23.03.23 07:34:16 (+0100)  sound-supervisor  Error connecting to audio block - Retry failed: connect ECONNREFUSED 192.168.2.183:4317
23.03.23 07:34:18 (+0100)  sound-supervisor  Error connecting to audio block - Retry failed: connect ECONNREFUSED 192.168.2.183:4317
23.03.23 07:34:22 (+0100)  sound-supervisor  Connected to PulseAudio at 192.168.2.183:4317
23.03.23 07:34:22 (+0100)  sound-supervisor  Server protocol version: 35
23.03.23 07:34:22 (+0100)  sound-supervisor  Client protocol version: 33
23.03.23 07:34:21 (+0100)  audio  W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
23.03.23 07:34:25 (+0100)  audio  E: [alsa-sink-bcm2835 Headphones] alsa-sink.c: ALSA woke us up to write new data to the device, but there was actually nothing to write.
23.03.23 07:34:25 (+0100)  audio  E: [alsa-sink-bcm2835 Headphones] alsa-sink.c: Most likely this is a bug in the ALSA driver 'snd_bcm2835'. Please report this issue to the ALSA developers.
23.03.23 07:34:25 (+0100)  audio  E: [alsa-sink-bcm2835 Headphones] alsa-sink.c: We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.

>

vescovd1, 2023-08-13:
same issue here. exact same set up.  No audio.
Please fix!

vescovd1, 2023-08-14:
audio  
 audio  --- Audio ---
 audio  Starting audio service with settings:
 audio  - pulseaudio 15.0
 audio  - Pulse log level: NOTICE
 audio  - Default output: RPI_HEADPHONES
 audio  - Default volume: 75%
 audio  
 audio  Detected audio cards:
 audio  0 vc4hdmi0 vc4-hdmi-vc4-hdmi-0
 audio  1 vc4hdmi1 vc4-hdmi-vc4-hdmi-1
 audio  
 audio  
 audio  WARNING: BCM2835 audio card not found, are you sure you are running on a Raspberry Pi?
 audio  Setting audio routing rules...
 audio  Routing 'balena-sound.input' to 'snapcast'.
 audio  Routing 'balena-sound.output' to '0'.
 audio  W: [pulseaudio] main.c: This program is not intended to be run as root (unless --system is specified).
 audio  E: [pulseaudio] module-alsa-card.c: Failed to find a working profile.
 audio  E: [pulseaudio] module.c: Failed to load module "module-alsa-card" (argument: "device_id="0" name="platform-fef00700.hdmi" card_name="alsa_card.platform-fef00700.hdmi" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=yes card_properties="module-udev-detect.discovered=1""): initialization failed.
 audio  W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
 audio  W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11

vescovd1, 2023-08-14:
fixed with https://forums.balena.io/t/raspberry-pi-4-no-3-5mm-sound-using-balenasound/356827