iotsound/iotsound #641: Hifiberry dacplushd

arjan5340, 2023-05-22:
Hello,

I have 3 Raspberry pi's 4 (64bit) On each PI i have a Hifiberry sound card "Hifiberry-dacplushd" Everything was good for almost 3 years. After doing an update i don't see my sourdcards anymore. 

i try really all variations of settings but noting helps. 

Please can anybody help me i read already hours on internet. 

[balenaOS 2.115.1](https://github.com/balena-os/meta-balena/blob/master/CHANGELOG.md)
CURRENT RELEASE [3.9.6](https://dashboard.balena-cloud.com/fleets/2054717/releases/2595738)

arjan5340, 2023-05-22:
Update. I have installed hifiberry OS and my hifiberry card work well.

johan489, 2023-09-19:
I have the same issue. The Hifiberry dacplus used to work but stopped after upgrade.

balenaOS 2.105.1+rev1
CURRENT RELEASE 3.9.3