iotsound/iotsound #649: No audio from Raspberry Pi 4 on 3.5mm jack

jeremybickerstaffe, 2023-11-09:
Raspberry Pi 4 with 8Gb RAM
balenaOS 4.0.26+rev1
No DAC

I installed balenasound and didn't change any settings.
I can connect to the device with any of Bluetooth / Airplay / Spotify Connect.

With AUDIO_OUTPUT set to the default of AUTO
Sound is only available over HDMI, can't get anything to play from 3.5mm jack

After changing AUDIO_OUPUT to RPI_HEADPHONES
No sound plays from the 3.5mm jack

I have done a hardware check that the 3.5mm jack is working correctly under Raspbian 

Suggestions please?

dotzeno, 2023-11-22:
https://forums.balena.io/t/raspberry-pi-4-no-3-5mm-sound-using-balenasound/356827

jeremybickerstaffe, 2023-11-24:
Thanks for the links @dotzeno
Removing the overlay fixed the issue for me
I had already searched but not found the right pages
What I did find was the balenasound support page which says for troubleshooting support to create an issue on Github 
https://balena-labs-projects.github.io/balena-sound/support#contact-us