iotsound/iotsound #652: Is this still being worked on?

C-Fu, 2023-12-16:
With the bluetooth unpair re-pair (as well as the bluetooth connection being rejected after the first few times using) bug still existing after months with no clear answer nor workaround, I have to ask if this project will ever be in a usable state? Or, which older version didn't have this issue?

Not trying to sound demanding (I'm not!), but if it's in a unusable state, shouldn't the project be declared as unusable (yet)?

bam80, 2024-07-27:
Is this?
I'm considering it but do not want to even try abandoned project.

seamusdemora, 2025-05-20:
Looks like another example of over-promise and under-deliver in RasPi-World  :( 

Ever wondered how many absolutely broken Linux projects there are on the Internet?  Seems to me the numbers are climbing... time for a new hobby - woodworking maybe?