iotsound/iotsound #658: Balena-sound docs are down, and an offer to help as a maintainer

MaximUltimatum, 2024-04-06:
Hello! I have noticed that as of today, April 6th, the docs for this project appear to be down.
https://balena-labs-projects.github.io/balena-sound/
The last tagged version of this project was in 2022, by @maggie44 

I dearly love this project. It has provided an affordable way to enjoy music as a family for me and my loved ones for several years now, and been a feature of gatherings of family and friends in my life.

Would an extra maintainer for this project be helpful? I'm thankful for what balena has made available to the community, and would be happy to give back in the form of development help.

I've been a microcontroller hobbyist for several years, and a (Kubernetes hobbyist as well)[https://github.com/MaximUltimatum/kube-homelab].

If an extra maintainer would be helpful to allow this project to continue to thrive, what would be most helpful? I can help triage issues, just try and stay on top of updates to the docs, or work on resolving old outstanding Issues and PRs.

@-ing existing folks who look like maintainers.
@maggie44 @thgreasi @alanb128

C-Fu, 2024-04-07:
just fork and move on, they don't seem to care about this. Doesn't seem to be in a working state atm anyways.

MaximUltimatum, 2024-04-09:
@vipulgupta2048 
I was reviewing  #654 and #655 and thought they might be related to the documentation suddenly breaking.
I can close #659 if you have a better solution? It was just meant as a patch to get the docs back up and running, since I wasn't sure why they broke in the first place.

alanb128, 2024-04-09:
@MaximUltimatum we're looking into some options to restore the documentation but it may take a little longer. Thanks for your patch, we'll let you know if we have a better solution.

vipulgupta2048, 2024-04-16:
Hello, I didn't want to ping before they went live but we are in the process of moving the docs to docusaurus at the moment. The docs did break because of the move to a new org. We also found that the current docs were built using Landr, and a migration to docusaurus was due. They are currently blocked due to a configuration issue that will soon be resolved. I already have them running locally on my system and will deploy them to production. 

[image]

vipulgupta2048, 2024-04-17:
Update: Deployed docusaurus docs to staging. Should be live soon: https://github.com/balena-io-experimental/balena-sound/pull/655#issuecomment-2060865643

bam80, 2024-07-27:
Status?

vipulgupta2048, 2024-07-29:
@bam80 Since April 17, the docs have been online here: https://balena-sound.pages.dev/

bam80, 2024-07-29:
So we can close this or anything left?

closed 2024-07-29 by vipulgupta2048