iotsound/iotsound #666: Librespot update to GoLang

jaragoncr99, 2024-07-31:
**Is your feature request related to a problem? Please describe.**
Is this project still alive? Will it get the new version of librespot?

https://github.com/devgianlu/go-librespot

**Describe the solution you'd like**
It seems like dev has stopped on the original librespot, and some of the bugs that cause restarts, not playing, etc will not be resolved without migration. 

**Describe alternatives you've considered**
N/A

**Additional context**
Add any other context or screenshots about the feature request here.

mivittek, 2024-10-27:
We currently upgraded librespot to version 0.5.0 in this thread #669

JaragonCR, 2025-11-15:
I've integrated golang librespot and submitted a pull to discuss it.  But

JaragonCR, 2026-03-03:
https://github.com/iotsound/iotsound/pull/700