iotsound/iotsound #669: Track skipping due to upstream Spotify API issues.

brettmillerb, 2024-09-04:
**Describe the problem**
When connecting to Balena raspberryPi device the tracks skip continuously.

**To Reproduce**
Open Spotify and attempt to connect to Balena Airplay device.
Tracks just begin skipping, some tracks play but the majority do not.

**Expected behavior**
Track selected plays on Balena device

**Screenshots**
```
 spotify  [2024-09-04T11:50:44Z ERROR librespot_core::channel] channel error: 2 0
 spotify  [2024-09-04T11:50:44Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
 spotify  [2024-09-04T11:50:44Z WARN  librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 11583470944873040899923648362564505491, audio_type: Track }>: ()
 spotify  [2024-09-04T11:50:44Z INFO  librespot_playback::player] Loading <ApÃ©ro Spritz> with Spotify URI <spotify:track:4K9CUUcvjacTnHdnAnzGRP>
```
Additional Errors:
```
 spotify  [2024-09-04T11:54:01Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
 spotify  [2024-09-04T11:54:01Z WARN  librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 297718726312721626386930062394132188552, audio_type: Track }>: ()
```

**Device information**
 - Board (hardware info): Raspberry Pi 3
 - Device type (as seen in balenaCloud): Raspberry Pi 3
 - balenaOS version: balenaOS 5.1.20+rev1

**Additional context**
Seems a Spotify API change is the cause and recommendation is to build librespot from dev branch which does not have the issue.

https://github.com/librespot-org/librespot/issues/1322

mivittek, 2024-09-04:
I can see in 
[/plugins/spotify/Dockerfile.template](https://github.com/balena-io-experimental/balena-sound/blob/b258569343327c0091fb5e03858eb12b50f37590/plugins/spotify/Dockerfile.template)

we are using image
`FROM andrewn/librespot:0.4.2-pulseaudio`

@andrewn is it possible for you to rereate docker image of librespot from dev branch ?

brettmillerb, 2024-09-05:
Because I am a masochist who doesn't know when to quite whilst he's ahead and I already possess a docker hub account I thought I'd try building the image myself and ran into a few issues:

Modified `build-pulseaudio.sh`

```bash
# not sure if this should align with the dev branch build but it was unique so 🤷 
# version = "0.5.0-dev" is dev branch version in cargo.toml
LIBRESPOT_VERSION="0.4.2-dev"
DOCKER_NAMESPACE="brettmillerit"
```
exec:
`bash build-pulseaudio.sh`

```bash
=> WARN: FromAsCasing: 'as' and 'FROM' keywords' casing do not match (line 4)                                                      0.1s
 => ERROR [internal] load metadata for docker.io/balenalib/rpi-alpine:edge                                                          0.9s
 => CANCELED [internal] load metadata for docker.io/balenalib/rpi-alpine:3.14-run 

ERROR: failed to solve: balenalib/rpi-alpine:edge: failed to resolve source metadata for docker.io/balenalib/rpi-alpine:edge: error getting credentials - err: exit status 255, out: ``
```
I then tried rebuilding using `:3.14` and received a cargo error (I don't remember why I thought changing from `edge` would work:

```bash
> [librespot-builder 6/6] RUN cargo build   --release   --features alsa-backend,pulseaudio-backend   --verbose:
0.997 error: failed to parse manifest at `/app/librespot-src/Cargo.toml`
0.998 
0.998 Caused by:
0.999   feature `edition2021` is required
0.999 
0.999   this Cargo does not support nightly features, but if you
0.999   switch to nightly channel you can add
0.999   `cargo-features = ["edition2021"]` to enable this feature
------
Dockerfile.pulseaudio.template:20
--------------------
  19 |     
  20 | >>> RUN cargo build \
  21 | >>>   --release \
  22 | >>>   --features alsa-backend,pulseaudio-backend \
  23 | >>>   --verbose
  24 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c cargo build   --release   --features alsa-backend,pulseaudio-backend   --verbose" did not complete successfully: exit code: 101
```

I moved back to `edge` and _I think_ I have it building....or it's stuck building and going to throw an error eventually 🤷

brettmillerb, 2024-09-06:
Went back to check the original issue and seems Spotify have reverted their API version changes

https://github.com/librespot-org/librespot/issues/1322#issuecomment-2334271629

However, when Spotify decide to make those changes Balena will need to utilise the new `0.5.0` and the container images will need to be rebuilt and updated.

Guess I learned a tonne about the project 🤣

andrewn, 2024-09-14:
Hi @brettmillerb, sorry for the delay. I can try and do a new build of the repo in the next couple of days.

I guess 0.5.0 is backwards compatible with the old API too?

brettmillerb, 2024-09-19:
Hi @andrewn 

> I guess 0.5.0 is backwards compatible with the old API too?
Spotify rolled back the changes and I am still using the dev branch build I generated.

I am way past the realms of my existing knowledge as it is. I have noticed with the `dev` branch build that there are random disconnects when selecting my raspi from Spotify.

May be worth hanging fire for an official release of 0.5.0 before rebuilding the docker images since Spotify rolled back the changes their end 🤷‍♂️

mivittek, 2024-10-20:
Hi there, currently, i am again getting a lot of:
```
librespot_core::session] Connection reset by peer (os error 104)
librespot_connect::spirc] subscription terminated
```

As suggested in docs, there is no actual solution, only to upgrade.
[balena-sound-spotify-connect-os-error-104](https://forums.balena.io/t/balena-sound-spotify-connect-os-error-104/360126/3)

There is already new librespot release [v0.5.0](/librespot-org/librespot/releases/tag/v0.5.0) avalible.

i forked balenasound and updated tag for docker image 
https://github.com/mivittek/balena-sound/blob/master/plugins/spotify/Dockerfile.template

and also andrewn librespot, where i bumed tag versions to 0.5.0
https://github.com/mivittek/librespot-docker

Bui i am unable to build docker image with github actions, there is some weird cargo build error.
https://github.com/mivittek/librespot-docker/actions/runs/11425163023/job/31786487840

andrewn, 2024-10-26:
It took a bit of fiddling around, but I have a version that seems to build v0.5.0 here: andrewn/librespot-docker#3

The GitHub actions have never worked, so I'm building them manually on my laptop.

The only thing is that I don't use balena-sound anymore so once they're up, someone's going to need to run them on a Pi to make sure they work.

I'll post here once they're on dockerhub.

mivittek, 2024-10-26:
Cool, i have pi4 and pi3 at my disposal, so i can try to build it on device :)

wheredidthecostgo, 2024-10-26:
> It took a bit of fiddling around, but I have a version that seems to build v0.5.0 here: [andrewn/librespot-docker#3](https://github.com/andrewn/librespot-docker/pull/3)
>
> The GitHub actions have [...]

I can give it a go on a pi2

andrewn, 2024-10-26:
I left my laptop cross-compiling all the binaries and went out for the day. 

I meant it would be great to check the compiled binaries work with balena-sound once they’re uploaded to docker.

andrewn, 2024-10-27:
@mivittek @wheredidthecostgo The new v0.5.0 images are now [published to docker hub as `andrewn/librespot`](https://hub.docker.com/r/andrewn/librespot/tags?name=0.5.0). 

Please test them out and let me know how they work for you

mivittek, 2024-10-27:
Well, i managed to build it and pull form docker repo, thanks for that. But there is error while running build. Some manifest missing ?

```
[airplay]           Step 1/10 : FROM mikebrady/shairport-sync:4.1.1 as shairport
[spotify]           Step 1/4 : FROM andrewn/librespot:0.5.0-pulseaudio
[spotify]           manifest for andrewn/librespot:0.5.0-pulseaudio not found: manifest unknown: manifest unknown
---- deleted some of build data

[Error]             Some services failed to build:
[Error]               Service: spotify
[Error]                 Error: manifest for andrewn/librespot:0.5.0-pulseaudio not found: manifest unknown: manifest unknown
[Info]              Built on df9c640
[Error]             Not deploying release.
```

i tried to debug it, but for main packge - without platform there is no manifest
```
docker manifest inspect andrewn/librespot:0.5.0-pulseaudio      
no such manifest: docker.io/andrewn/librespot:0.5.0-pulseaudio
```

When i edit it manually as you cen see it in commit, it works as a charm.
[/mivittek/balena-sound/commit/1847d9f974714fd421f168e48bace59701ccfd92](https://github.com/mivittek/balena-sound/commit/1847d9f974714fd421f168e48bace59701ccfd92)

andrewn, 2024-10-27:
@mivittek Thanks for reporting that!

I forgot to enable the "create manifest" line in the script. 🤦 

If you try again it should work.

wheredidthecostgo, 2024-10-27:
> @mivittek @wheredidthecostgo The new v0.5.0 images are now [published to docker hub as `andrewn/librespot`](https://hub.docker.com/r/andrewn/librespot/tags?name=0.5.0).
> 
> Please test them out and let me know how they work for you

I've built and deployed!   I'll test and see if it corrects the (os error 104)

mivittek, 2024-10-27:
I redeployed image and since [commit/f954c98](https://github.com/mivittek/balena-sound/commit/f954c986cbd9f5b665fdf76ffa1ee4cb2702a4fc) I am running 4+ hours and no error yet, so I got high hopes :)

wheredidthecostgo, 2024-10-27:
@andrewn  It has worked perfect for me all day!  Thank you!!!

mivittek, 2024-11-01:
Its still working like a charm, should we create PR with update + bumped release version ?

wheredidthecostgo, 2024-11-02:
It worked fine for a while, now I'm getting status code 502 errors.  See below.  I'm guessing it is some type of bad gateway error, but not sure why it would be happening.  

But os 104 errors are gone, so I'm not sure what to do.  

[image]

mivittek, 2024-11-02:
Hi there !
 Error 502 happens after starting playing after some time without playback. It happens for me as well, but it does not 

Log: 
```
Supervisor starting
 spotify  [2024-11-02T17:13:30Z ERROR librespot_connect::spirc] ContextError: Error { kind: Unknown, error: StatusCode(502) }
```
This issue is documented in librespot , without any real solution :(
Comments in this tread are saying that session handling has bee n imporved and ther eis new librespot version, 0.6.0

reference:
https://github.com/librespot-org/librespot/discussions/609

wheredidthecostgo, 2024-11-05:
Thanks!  

> Hi there !
>  Error 502 happens after starting playing after some time without playback. It happens for me as well, but it does not 
> 
> Log: 
> ```
> Supervisor starting
>  spotify  [2024-11-02T17:13:30Z ERROR librespot_connect::spirc] ContextError: Error { kind: Unknown, error: StatusCode(502) }
> ```
> This issue is documented in librespot , without any real solution :(
> Comments in this tread are saying that session handling has been  improved and there is new librespot version, 0.6.0
> 
> reference:
> https://github.com/librespot-org/librespot/discussions/609

JaragonCR, 2025-08-14:
I started trying golang librespot, is there any appetite to try this? Currently is a rudimentary implementation on my Pi4 but it is working, had to revert it to use ALSA instead of Pulse, but the rest seems to work just fine, very fast and snappy. 

https://github.com/JaragonCR/iotsound/blob/librespot/plugins/librespot/Dockerfile.template

Should we do more testing in this area and build the docker images?