iotsound/iotsound #671: Spotify Login Failing

tlewis99, 2024-10-20:
Starting to get this error on two different fleets using three different confirmed Spotify accounts.  I wonder if this has anything to do with a change to spotify that seems to push auth from thier app to a browser and back recently.  

spotify  [2024-10-20T15:54:26Z ERROR librespot] Connection failed: Login failed with reason: Bad credentials

tlewis99, 2024-10-20:
Should have said release 3.10.0 and 3.9.3

tlewis99, 2024-11-15:
Also both fleets are raspberry pie 4’s

Bananz0, 2025-02-18:
I'm also experiencing this on a Pi4B
` spotify  Starting Spotify plugin...
 spotify  Device name: Harman Kardon Soundsticks III
 spotify  Volume normalization disabled.
 spotify  Spotify audio cache enabled.
 spotify  [2025-02-18T05:15:47Z INFO  librespot] librespot 0.4.2 UNKNOWN (Built on 2022-08-01, Build ID: leSkOw9g, Profile: release)
 spotify  [2025-02-18T05:15:47Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gue1.spotify.com:4070
 spotify  [2025-02-18T05:15:47Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:443
 spotify  [2025-02-18T05:15:47Z INFO  librespot_core::session] Connecting to AP "ap-gew1.spotify.com:4070"
 spotify  [2025-02-18T05:15:47Z ERROR librespot] Connection failed: Login failed with reason: Bad credentials
Service exited 'spotify sha256:5f69f467e03996d60b355472e030959ce49933eda7c66afa5bd6192490fea30c'
Restarting service 'spotify sha256:5f69f467e03996d60b355472e030959ce49933eda7c66afa5bd6192490fea30c'
`

JaragonCR, 2026-03-03:
@tlewis99 are you still using this?