iotsound/iotsound #679: Balena Sound Fleet

rogerdelvin, 2024-12-27:
Hello,

I am in the process of creating a Balena-Sound fleet for a multi-room audio setup to listen to Spotify.

Device 1 - RPI 4
Device 2 - RPI 5

Both devices have been added to my balena-sound fleet. I can also see both devices on my Spotify app. However, there is no audio output on either. 

Not sure what is wrong or if I have missed something.

I am an absolute noob to RPI and Balena! Please help.

regards,

Roger

AkshayChikhalkar, 2025-01-24:
Hi Roger,

I am facing a problem flashing the proper balena sound and would like to connect.
If it is ok with you, could you share where we can connect.

Thank you!

theoisadoor, 2025-01-31:
@rogerdelvin I'm not sure that RPI 5 is technically supported, but have you attached a DAC to the RPI 5 due to the lack of 3.5mm jack? If so you will need to add the relevant `BALENA_HOST_CONFIG_dtoverlay` (and therefore potentially `BALENA_HOST_CONFIG_gpio`) parameters into your configuration, and Variable set to `AUDIO_OUTPUT = DAC`, which can be found here: `docs/05-audio-interfaces.md`. If you check #670 you will see updated documentation for this.
Example provided for my devices, which use Pimoroni Pirate Audio SHIMs. I have set my configuration fleet wide since they're all the same device, so set yours per device:

`Variables`
[image]
`Configuration > Custom Configuration`
[image]

As for your RPI 4, you will most likely set `AUDIO_OUTPUT = RPI_AUTO`, unless you want to specify `AUDIO_OUTPUT = RPI_HEADPHONES` (docs found here: `docs/03-customization.md`).

It also seems that you may have set your devices to `STANDALONE` since you can see them both individually - you should only see one for multi-room. One device needs to act as the server, and all others need to be clients. The device set as the server will be the only name you see in Spotify, which can be customised with `SOUND_DEVICE_NAME` (seen below). You need the following:
Device Variable set to `SOUND_MODE = MULTI_ROOM` for the 'server', and `SOUND_MODE = MULTI_ROOM_CLIENT` for all other devices in the fleet.

[image]

I hope this helps.

rogerdelvin, 2025-02-01:
> Hi Roger,
> 
> I am facing a problem flashing the proper balena sound and would like to connect. If it is ok with you, could you share where we can connect.
> 
> Thank you!

Hi Akshay,

There are plenty of YouTube videos on how to flash balena cloud. Its quite simple and straightforward.

Good day!

rogerdelvin, 2025-02-01:
> [@rogerdelvin](https://github.com/rogerdelvin) I'm not sure that RPI 5 is technically supported, but have you attached a DAC to the RPI 5 due to the lack of 3.5mm jack? If so you will need to add the relevant `BALENA_HOST_CONFIG_dtoverlay` (and therefore potentially `BALENA_HOST_CONFIG_gpio`) parameters into your configuration, and Variable set to `AUDIO_OUTPUT = DAC`, which can be found here: `docs/05-audio-interfaces.md`. If you check [#670](https://github.com/balena-io-experimental/balena-sound/pull/670) you will see updated documentation for this. Example provided for my devices, which use [Audio DAC SHIMs by Pimoroni](https://shop.pimoroni.com/products/audio-dac-shim-line-out?variant=32343184965715). I have set my configuration fleet wide since they're all the same device, so set yours per device:
> 
> `Variables` [image] `Configuration > Custom Configuration` [image]
> 
> As for your RPI 4, you will most likely set `AUDIO_OUTPUT = RPI_AUTO`, unless you want to specify `AUDIO_OUTPUT = RPI_HEADPHONES` (docs found here: `docs/03-customization.md`).
> 
> It also seems that you may have set your devices to `STANDALONE` since you can see them both individually - you should only see one for multi-room. One device needs to act as the server, and all others need to be clients. The device set as the server will be the only name you see in Spotify, which can be customised with `SOUND_DEVICE_NAME` (seen below). You need the following: Device Variable set to `SOUND_MODE = MULTI_ROOM` for the 'server', and `SOUND_MODE = MULTI_ROOM_CLIENT` for all other devices in the fleet.
> 
> [image]
> 
> I hope this helps.

Hello !

Many thanks for checking out my issue and your detailed response.

Yes, I have any ifi Audio DAC on my RPI5. 

AUDIO_OUTPUT : AUTO
(on both RPIs)

I found a solution on an other thread and made the changes - removed "vc4-kms-v3d" overlay option from from Define DT overlays under device configuration variables. Its working after that. 

https://forums.balena.io/t/raspberry-pi-4-no-3-5mm-sound-using-balenasound/356827
https://forums.balena.io/t/raspberry-pi4-no-audio/372629

Though I have not set the SOUND_MODE = MULTI_ROOM for the Server / Client, i think Balena sound is automatically setting one device as a master and the others the client within the fleet.

Good day !

theoisadoor, 2025-02-01:
> Many thanks for checking out my issue and your detailed response.

My pleasure.

> Yes, I have any ifi Audio DAC on my RPI5.
> 
> AUDIO_OUTPUT : AUTO
> (on both RPIs)

That makes sense, since you're using a USB DAC, which uses PulseAudio.

> I found a solution on an other thread and made the changes - removed "vc4-kms-v3d" overlay option from from Define DT overlays under device configuration variables. Its working after that.

Glad to hear it was resolved! It is also worth creating a PR updating the docs to reflect this requirement to remove that configuration for devices running 64bit balena.

> Though I have not set the SOUND_MODE = MULTI_ROOM for the Server / Client, i think Balena sound is automatically setting one device as a master and the others the client within the fleet.

That is correct, `MULTI_ROOM` is the default variable, and it does automatically choose which device to be the master vs slave. It doesn't explain why you are (or were) seeing the two devices separately however, as they should only come up under one name in `MULTI_ROOM`, as defined by `SOUND_DEVICE_NAME` as I mentioned previously. Is it still exhibiting this behaviour?

rogerdelvin, 2025-02-01:
> My pleasure.
>
>
> That makes sense, since you're using a USB DAC, which uses PulseAudio.
>
>
> Glad to hear it was resolved! It is also worth creating a PR updating the docs to reflect this requirement to [...]

I do see 2 devices on my Spotify, however when I select one for the Output, the other device becomes the slave I guess in the Multi-room setup

closed 2025-02-13 by rogerdelvin