iotsound/iotsound #682: Balena Sound NO AUDIO Rpi4b Release 3.10 on OS 6.3.18

flatbattery, 2025-02-12:
Balena Sound brand new Rpi4b Release 3.10 on OS 6.3.18 installed using DEPLOY button and NO modifications from the defaults.

I have just bought a new RPi4b specifically for Balena Sound. I have used the DEPLOY button and chosen the RPi4b option. I am using wired ethernet. Spotify finds and ‘plays’ but no sound is output from the 3.5mm jack. Spotify was started via network (app on phone). Also no sound output if I connect from phone via Bluetooth. I could try via the RPi HDMI port as I have an HDMI audio splitter. I will post results of that soon. I have looked at other posts but they are from 1 or 2 years ago so I would rather have help with it while it is still in the ‘default’ state

Spotify does work with my VolumeIO setup.

Is anyone able to offer any help before I change anything as the ‘default’ deployment to a standard RPi would seem to ‘work out of the box’
Thanks

**Expected behavior**
From Balena Sound notes 'it should work with default setup if the 'deploy button' is used

**Screenshots**
NA

**Device information**
 - Board (hardware info): [e.g. Raspberry Pi 4B]
 - Device type (as seen in balenaCloud): [e.g. Raspberry Pi 4 (using 64bit OS)]
 - balenaOS version: [e.g. balenaOS v6.3.18]
 - balena sound release 3.10

**Additional context**
Spotify 'plays' from the PC or from phone app but no sound is output from the 3.5mm jack

M-Kwapil, 2025-03-03:
I was having a similar issue and this fixed it for me.
Removing "vc4-kms-v3d" overlay option from from Define DT overlays under device configuration variables as per [GH](https://github.com/balenalabs/balena-sound/issues/82) issue.

Kullsyreholdig, 2025-04-19:
I had the same problem, tried a lot of things but what made it work for me was:

AUDIO_OUTPUT: RPI_AUTO
RPI_HEADPHONES did not work.

RESIN_HOST_CONFIG_dtoverlay="cma-320" aka Device Tree Overlay / Define DT overlays

This worked on RPI 4B(master) and RPI 3b+(client)