iotsound/iotsound #684: No audio output on Pi4B with  dt-overlay "vc4-fkms-v3d" and on balenaOS 6.5.1+rev2 with Supervisor version 16.12.9

Bananz0, 2025-03-26:
There is no sound output on the RPi4B after updating the supervisor and balenaOS to 16.12.9 and 6.5.1+rev2 respectively. There was an issue before where I had no sound but fixed that by changing the dt overlay to `vc4-fkms-v3d` which fixed it and has been running for weeks. #82 
Updating the OS and supervisor made an issue where I can only fix it using 
`pactl load-module module-loopback latency_msec=50 source=balena-sound.output.monitor sink=alsa_output.platform-fe00b840.mailbox.stereo-fallback 26`
I have to do that every reboot. 

I get this on the audio console

` audio  E: [alsa-sink-bcm2835 Headphones] alsa-sink.c: ALSA woke us up to write new data to the device, but there was actually nothing to write.
 audio  E: [alsa-sink-bcm2835 Headphones] alsa-sink.c: Most likely this is a bug in the ALSA driver '(null)'. Please report this issue to the ALSA developers.
 audio  E: [alsa-sink-bcm2835 Headphones] alsa-sink.c: We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
`

**To Reproduce**
Steps to reproduce the behavior:
1. Set DT Overlay to `vc4-fkms-v3d` 
2. Update balenaOS
3. Update supervisor
4. Listen to the void

**Expected behavior**
Listen to Katy Perry and Led Zepplin

**Device information**
 - Board (hardware info):  Raspberry Pi 3B
 - Device type: Raspberry Pi 4 (using 64bit OS)
 - balenaOS version: balenaOS 6.5.1+rev2 ]
 - Device Tree Overlay: `vc4-fkms-v3d`

wout, 2025-04-13:
I think I'm experiencing the same problem: https://forums.balena.io/t/no-sound-output-on-raspberry-pi-4-with-balena-sound/374020

Could it have something to do with the sink name? In my case, the expected sink (`alsa_output.bcm2835-jack.stereo-fallback`) is actually called `alsa_output.Headphones.stereo-fallback`, although I have no idea how or where to change that value.

Kullsyreholdig, 2025-04-19:
I had the same problem, tried a lot of things but what made it work for me was:

`AUDIO_OUTPUT: RPI_AUTO`
RPI_HEADPHONES did not work.

`RESIN_HOST_CONFIG_dtoverlay="cma-320"` aka Device Tree Overlay / Define DT overlays

This worked on RPI 4B(master) and RPI 3b+(client)

JaragonCR, 2026-03-03:
@Kullsyreholdig  @wout @Bananz0  are you still using this project? Could you test this build?

https://github.com/JaragonCR/iotsound

Bananz0, 2026-09-04:
Apologies this issue made me switch to home assistant with Spotify headless on it!